HSU-ANT/ACME.jl #17: No Unitful support

freemin7, 2020-02-24:
Unitful.jl is a nice library which allows type inference over physical units making unit conversion errors less likely.
I tried the following:
`circ = @circuit begin
                j_in = voltagesource(9u"V")
                j_out = voltageprobe()
                j_in[+] == j_out[+]
                j_in[-] == j_out[-]
              end
`
You might want to look into Unitful and whether supporting it makes sense for you.

martinholters, 2020-02-24:
Agree this would be super nice to support. Unfortunately it's not just writing the code in a way that just happens to work with unitful numbers. Internally, it's quite likely that quantities of different units are collected in the same vector, but for performance, we definitely want that vector elements to be concretely typed, so that leaves dimensionless as best option _internally_.

The surface, however, would definitely benefit from supporting units. It's just that I'm reluctant to add dependencies. But if we get something like https://github.com/JuliaLang/Pkg.jl/issues/1285 eventually, I'd definitely like to add Unitful as an optional dependency.

bhalonen, 2021-07-31:
Will also note that supporting Unitful APIs makes life much easier to get started. Have lost weeks of work putting the wrong unit into a complex program I did not understand.

PR #93 (2023-01-26): Support Unitful in element construction
For example:
```julia
julia> using ACME, Unitful

julia> resistor(4.7e3) == resistor(4.7u"kΩ")
true
```
Note that this requires Julia 1.9 as it uses a package extension.

Closes #17.
closed 2023-01-26 by martinholters