HSU-ANT/ACME.jl #29: Triode / Pentode Models

dorpxam, 2020-05-28:
@martinholters , I've take a look to the differents triode/pentode/... models used in the scientific litterature, I see you take part in differents study. For example the 2011 paper : "_A Triode Model for Guitar Amplifier Simulation with Individual Parameter Fitting_" with Kristjan Dempwolf and Udo Zölzer.

Can I ask you if you can orient me to the current _state of the art_ for this subject? You may be advised on the subject. If not, no problem, I will take few weeks to work on the subject.

martinholters, 2020-05-28:
Although I've been somewhat involved with Kristjan Dempwolfs work, I'm not sure about my qualification concerning the current state of the art. I'm not aware of any significant publications on the subject since then, but I may easily have missed some.

If I remember correctly, we were quite happy with our results on the triode, so I would probably have started with that model. But I'm obviously biased here :smile: We also did some work on pentodes, but as far as I remember, results were a bit mixed there. It might even make sense to implement different models to facilitate further research in this area. With the obvious downside that requires, well, implementing different models.

dorpxam, 2020-05-28:
So I will take probably few days before make a little feedback with the idea, and maybe a little help from you to validate my work ;)

cristianfarias, 2021-03-03:
Hi Martin,

I added the next triode element (COHEN AND HELIE model)  but it seems to not resolve successfully the initial conditions ?
1) First I add the next code in the file elements.jl

export resistor, potentiometer, capacitor, inductor, transformer,
       voltagesource, currentsource,
       voltageprobe, currentprobe, diode, bjt, mosfet, opamp, triode

...
...

@doc raw"""
    triode(;Gk=2.14e-3, uk=100.8, Ek=1.303, Ck=3.04, Gg=6.06e-4, Eg=1.354, Cg=13.9)

Creates a triode model

Pins: `plate`, `grid`, `cathode`
""" function triode(;Gk=2.14e-3, uk=100.8, Ek=1.303, Ck=3.04, Gg=6.06e-4, Eg=1.354, Cg=13.9)

    nonlinear_eq =
        @inline function (q)
            VgkT1, VpkT1, IgT1, IpT1 = q
            IgT1=Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg
            IpT1=Gk*(log(1+exp(Ck*(VpkT1/uk+VgkT1)))*1/Ck)^Ek-Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg
			dIgT1_dVgkT1=(1891999529292056969*exp((139*VgkT1)/10)*((10*log(exp((139*VgkT1)/10) + 1))/139)^(177/500))/(2305843009213693952000*(exp((139*VgkT1)/10) + 1))
			dIgT1_dVpkT1=0
			dIpT1_dVgkT1=(139421*exp((76*VgkT1)/25 + (19*VpkT1)/630)*((25*log(exp((76*VgkT1)/25 + (19*VpkT1)/630) + 1))/76)^(303/1000))/(50000000*(exp((76*VgkT1)/25 + (19*VpkT1)/630) + 1)) - (1891999529292056969*exp((139*VgkT1)/10)*((10*log(exp((139*VgkT1)/10) + 1))/139)^(177/500))/(2305843009213693952000*(exp((139*VgkT1)/10) + 1))
			dIpT1_dVpkT1=(139421*exp((76*VgkT1)/25 + (19*VpkT1)/630)*((25*log(exp((76*VgkT1)/25 + (19*VpkT1)/630) + 1))/76)^(303/1000))/(5040000000*(exp((76*VgkT1)/25 + (19*VpkT1)/630) + 1))
            res = @SVector [IgT1, IpT1]
            J = @SMatrix [dIgT1_dVgkT1 dIgT1_dVpkT1 -1.0 0.0;
                          dIpT1_dVgkT1 dIpT1_dVpkT1 0.0  -1.0]
            return (res, J)
        end
    return Element(mv=[1 0; 0 1; 0 0; 0 0],
                   mi = [0 0; 0 0; 1 0; 0 1],
                   mq = [-1 0 0 0; 0 -1 0 0; 0 0 -1 0; 0 0 0 -1], nonlinear_eq = nonlinear_eq,
                   ports = [:grid => :cathode, :plate => :cathode])
end

2) When I tried to calculate the model I get the next error 

julia> model = DiscreteModel(circ, 1/44100)
ERROR: Failed to find initial solution
Stacktrace:
 [1] error(::String) at .\error.jl:33
 [2] initial_solution(::Function, ::Array{Float64,1}, ::Int64) at C:\Users\cti4121\.julia\packages\ACME\bLRnf\src\ACME.jl:436
 [3] DiscreteModel(::Circuit, ::Float64, ::Type{HomotopySolver{CachingSolver{SimpleSolver}}}; decompose_nonlinearity::Bool) at C:\Users\cti4121\.julia\packages\ACME\bLRnf\src\ACME.jl:198
 [4] DiscreteModel at C:\Users\cti4121\.julia\packages\ACME\bLRnf\src\ACME.jl:150 [inlined] (repeats 2 times)
 [5] top-level scope at REPL[5]:1

cristianfarias, 2021-03-03:
before that, I define this circuit to simulate

vol=0.5;

circ = @circuit begin
    j3 = voltagesource(250), [-] ⟷ gnd, [+] ⟷ vcc # 250V power supply
    j1 = voltagesource(), [-] ⟷ gnd # input
    c1 = capacitor(47e-9), [1] ⟷ j1[+]
    r1 = resistor(1e6), [1] ⟷ c1[2], [2] ⟷ gnd
    r2 = resistor(70e3), [1] ⟷ c1[2]
    c2 = capacitor(2.5e-9), [1] ⟷ r2[2]
    t1 = triode(),
         [grid] ⟷ r2[2], [plate] ⟷ c2[2]
    r3 = resistor(100e3), [1] ⟷ vcc, [2] ⟷ t1[plate]
    r4 = resistor(1.5e3), [1] ⟷ t1[cathode], [2] ⟷ gnd
    c3 = capacitor(25e-6), [1] ⟷ t1[cathode], [2] ⟷ gnd
    c4 = capacitor(1e-6), [1] ⟷ t1[plate]
    p1 = potentiometer(100e3, (vol == nothing ? () : (vol,))...), [1] ⟷ gnd, [3] ⟷ c4[2]
    j2 = voltageprobe(), [-] ⟷ gnd, [+] ⟷ p1[2]# output
end

Regards

martinholters, 2021-03-09:
From a brief look, I think you'd want something like
```julia
res1 = Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg - IgT1
res2 = Gk*(log(1+exp(Ck*(VpkT1/uk+VgkT1)))*1/Ck)^Ek-Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg - IpT1
# ...
res = @SVector [res1, res]
```
I.e., the `res` vector has to hold the difference between the currents computed from the voltages and the currents provided as `q[3]` and `q[4]`. For convergence, it important that the Jacobian `J` is correct, which I cannot judge from a brief look, but given the strange numbers occurring in there, I assume you used some special tool to compute the derivatives for specific parameter values (probably the defaults)?

cristianfarias, 2021-03-10:
Hi Martin,

For example, the original equations than formulate the currents in the KOHEN AND HELIE model of a triode are the next:

IgT1=Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg

IpT1=Gk*(log(1+exp(Ck*(VpkT1/uk+VgkT1)))*1/Ck)^Ek-Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg

Using MATLAB I calculate the first derivatives from IgT1 and IpT1 respect to Vgk and Vpk , respectively

that I named in my code as follow   ---- >  dIgT1_dVgkT1
                                                                     dIgT1_dVpkT1
                                                                      dIpT1_dVgkT1
                                                                     dIpT1_dVpkT1

Here are the command that I use in MATLAB to calculate them

syms VgkT1 VpkT1 Gg Cg Eg  Gk Ck uk Ek;
IgT1=Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg;
IpT1=Gk*(log(1+exp(Ck*(VpkT1/uk+VgkT1)))*1/Ck)^Ek-Gg*(log(1+exp(Cg*VgkT1))*1/Cg)^Eg;
diff(IgT1,VgkT1)

  ans =

    (Eg*Gg*exp(Cg*VgkT1)*(log(exp(Cg*VgkT1) + 1)/Cg)^(Eg - 1))/(exp(Cg*VgkT1) + 1)

diff(IgT1,VpkT1)

  ans =

    0

diff(IpT1,VgkT1)

  ans =

    (Ek*Gk*exp(Ck*(VgkT1 + VpkT1/uk))*(log(exp(Ck*(VgkT1 + VpkT1/uk)) + 1)/Ck)^(Ek - 1))/(exp(Ck*(VgkT1 + VpkT1/uk)) + 1) - (Eg*Gg*exp(Cg*VgkT1)*(log(exp(Cg*VgkT1) + 1)/Cg)^(Eg - 1))/(exp(Cg*VgkT1) + 1)

diff(IpT1,VpkT1)

  ans =

    (Ek*Gk*exp(Ck*(VgkT1 + VpkT1/uk))*(log(exp(Ck*(VgkT1 + VpkT1/uk)) + 1)/Ck)^(Ek - 1))/(uk*(exp(Ck*(VgkT1 + VpkT1/uk)) + 1))

Now, when a tried to simulate the circuit with two 12AX7 triodes known as "Fender Type Guitar Preamp", and using the KOHEN AND HELLIE model ----- > then there isn't any error in the commands, but the simulated output there isn't to be the expected ???

vol6=0.5;
vol7=0.5;
vol8=0.5;
vol9=0.5;
vol10=0.1;

circ = @circuit begin
    j3 = voltagesource(385), [-] ⟷ gnd, [+] ⟷ vcc # 385V power supply
    j1 = voltagesource(), [-] ⟷ gnd # input
    r1 = resistor(32e3), [1] ⟷ j1[+]
    r2 = resistor(1e6), [1] ⟷ r1[2], [2] ⟷ gnd
    t1 = triode_koh(;Gk=2.14e-3, uk=100.8, Ek=1.303, Ck=3.04, Gg=6.06e-4, Eg=1.354, Cg=13.9),
         [grid] ⟷ r1[2]
    r3 = resistor(1.5e3), [1] ⟷ t1[cathode], [2] ⟷ gnd
    c1 = capacitor(22e-6), [1] ⟷ t1[cathode], [2] ⟷ gnd
    r4 = resistor(100e3), [1] ⟷ vcc, [2] ⟷ t1[plate]
    r5 = resistor(100e3), [1] ⟷ t1[plate]
    c2 = capacitor(250e-12), [1] ⟷ t1[plate]
    c3 = capacitor(100e-9), [1] ⟷ r5[2]
    c4 = capacitor(22e-9), [1] ⟷ r5[2]
    p6 = potentiometer(250e3, (vol6 == nothing ? () : (vol6,))...), [1] ⟷ c3[2], [3] ⟷ c2[2]
    p7 = potentiometer(250e3, (vol7 == nothing ? () : (vol7,))...), [1] ⟷ c4[2], [3] ⟷ c3[2]
    p8 = potentiometer(10e3, (vol8 == nothing ? () : (vol8,))...), [1] ⟷ gnd, [3] ⟷ c4[2]
    p9 = potentiometer(1e6, (vol9 == nothing ? () : (vol9,))...), [1] ⟷ gnd, [3] ⟷ p6[2]
    c5 = capacitor(120e-12), [1] ⟷ p6[2], [2] ⟷ p9[2]
    t2 = triode_koh(;Gk=2.14e-3, uk=100.8, Ek=1.303, Ck=3.04, Gg=6.06e-4, Eg=1.354, Cg=13.9),
         [grid] ⟷ p9[2]
    r10 = resistor(1640), [1] ⟷ t2[cathode], [2] ⟷ gnd
    c6 = capacitor(22e-6), [1] ⟷ t2[cathode], [2] ⟷ gnd
    r4 = resistor(100e3), [1] ⟷ vcc, [2] ⟷ t2[plate]
    c7 = capacitor(1e-6), [1] ⟷ t2[plate]
    p10 = potentiometer(400e3, (vol10 == nothing ? () : (vol10,))...), [1] ⟷ gnd, [3] ⟷ c7[2]
    j2 = voltageprobe(), [-] ⟷ gnd, [+] ⟷ p10[2]# output
end

connect!(circ, (:p7, :2), (:p7, :3))
connect!(circ, (:p8, :2), (:p8, :3))

model = DiscreteModel(circ, 1/44100)

y = run!(model, [0.5*sin(2π*1000/44100*n) for c in 1:1, n in 0:44099])
in=[0.5*sin(2π*1000/44100*n) for c in 1:1, n in 0:44099]
x=[n for c in 1:1, n in 0:44099]
z=transpose(y)
plot(x[40000:40400],z[40000:40400])
plot!(x[40000:40400],in[40000:40400])

Regards !!

cristianfarias, 2021-07-24:
> Sorry, your latest posts somehow slipped through unnoticed. Probably a bit late now, but in your example circuit, there are two r4's. Renaming the latter to, say, r11, yields plausible results. (Using the same designator twice should probably give a warning to help avoid such pitfalls).

Thanks Martin for your help.

I rename this variable and the output was the expected !!!

Regards

martinholters, 2021-08-18:
FWIW, with the recently released v0.9.5, the buggy circuit description above now emits
```
┌ Warning: redefinition of `r4`
```