HSU-ANT/ACME.jl #30: A GUI for ACME

dorpxam, 2020-05-28:
I've remove this part from my previous message, out-of-subject. 

What do you think about that? 

Using a crossplatform toolkit like JUCE, a runtime wrapper to Julia's code, maybe a JIT code generator (LLVM) for very fast instant simulation. ACME can become more attractive, not only for pure scientist, but for guitar oriented circuit manufactors.

I keep in mind LiveSPICE that is a very cool but dead-project : https://www.livespice.org/

martinholters, 2020-05-28:
Oh sure, a GUI where you draw a circuit, push a button and after a bit of compilation time get a plugin for your favorite DAW would be fantastic. However, I think that this project—ACME.jl—should stay focused on turning a circuit description into a model with some methods to actually run the model, if maybe not fully optimized. That is, I can very well see this as a backend for a GUI developed elsewhere. And I'd happily incorporate changes here to make that feasible (a proper API for extracting the model description comes to mind).

So while I'm supportive of the idea, I'll probably close this issue eventually simply because I don't want the GUI to be part of this project. Ideally, by that time I can point to a new project where the GUI resides. Meanwhile, I'll leave this open for a bit to be more inviting to any discussion on the topic.

dorpxam, 2020-05-28:
Ofcourse, the basic idea is that kind of project would be a third party independant project. For the moment, my current project is not really connected to any part of ACME, I focus to a cleanest way to draw a circuit with basic system for kind of MNA 

You can close this issue. I will probably open the project soon to my github.

Thank you for the answer

DhruvaSambrani, 2021-01-03:
Do update if you do so!

closed 2021-08-02 by martinholters