HSU-ANT/ACME.jl #49: Combining the two op amp models

sbutron, 2021-04-20:
Is it possible to combine the two op amp models? Basically I'm trying define a gain and GBP for an op amp as well as setting a max output voltage limit.

martinholters, 2021-04-21:
There is not a single model that combines limited GBP and output saturation, but you can put the two existing models in series to obtain what you. E.g. use a linear model with the desired maximum gain and GBP followed by the saturating model with (small-signal) gain equal to one. Alternatively, you could first apply the saturation and then the lowpass (GBP limiting). I'm not sure which of these alternatives better resembles what an actual op amp does. (That's the main reason I haven't implemented a combined model, yet.)

sbutron, 2021-04-23:
Thanks, I'll give that a try!