HSU-ANT/ACME.jl #82: test fails in Julia 1.7.2

sadguitarius, 2022-04-03:
Hi, I'm new to Julia and trying to get ACME.jl working. I'm running into an error that seems to be related to Julia 1.7.2 breaking something, but I'm not proficient enough in the language yet to figure out what's going on. I've attached the output of running the package tests below. Is there a way to get the package working with the current version of Julia? I think handling multiple Julia versions is too much for my poor brain to handle right now. Thanks!

```
Test Summary: | Pass  Total
topomat       |    3      3
Test Summary: | Pass  Total
LinearSolver  |   10     10
resistor-diode: Error During Test at C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\test\runtests.jl:61
  Got exception outside of a @test
  TypeError: in <:, expected Type, got Vararg
  Stacktrace:
    [1] #s10#5
      @ C:\Users\sadguitarius\.julia\packages\StaticArrays\DBECI\src\util.jl:20 [inlined]
    [2] var"#s10#5"(T::Any, ::Any, #unused#::Any)
      @ StaticArrays .\none:0
    [3] (::Core.GeneratedFunctionStub)(::Any, ::Vararg{Any})
      @ Core .\boot.jl:580
    [4] macro expansion
      @ C:\Users\sadguitarius\.julia\packages\StaticArrays\DBECI\src\SMatrix.jl:35 [inlined]
    [5] SArray
      @ C:\Users\sadguitarius\.julia\packages\StaticArrays\DBECI\src\SMatrix.jl:32 [inlined]
    [6] #23
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\elements.jl:233 [inlined]
    [7] #72
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\circuit.jl:82 [inlined]
    [8] _apply_all
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\circuit.jl:16 [inlined]
    [9] CircuitNLFunc
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\circuit.jl:12 [inlined]
   [10] #106
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\ACME.jl:181 [inlined]
   [11] #108
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\ACME.jl:192 [inlined]
   [12] evaluate!(nleq::ACME.ParametricNonLinEq{ACME.var"#108#122"{Matrix{Float64}, ACME.var"#106#120"{Vector{Float64}, ACME.CircuitNLFunc{Tuple{ACME.var"#72#73"{StaticArrays.SVector{2, Int64}, ACME.var"#23#24"{Float64, Int64}}}}}}, typeof(ACME.default_set_p), typeof(ACME.default_calc_Jp), Tuple{Vector{Float64}, Matrix{Float64}}}, z::Vector{Float64})
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\solvers.jl:36
   [13] set_extrapolation_origin(solver::SimpleSolver{ACME.ParametricNonLinEq{ACME.var"#108#122"{Matrix{Float64}, ACME.var"#106#120"{Vector{Float64}, ACME.CircuitNLFunc{Tuple{ACME.var"#72#73"{StaticArrays.SVector{2, Int64}, ACME.var"#23#24"{Float64, Int64}}}}}}, typeof(ACME.default_set_p), typeof(ACME.default_calc_Jp), Tuple{Vector{Float64}, Matrix{Float64}}}}, p::Vector{Float64}, z::Vector{Float64})
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\solvers.jl:184
   [14] SimpleSolver(nleq::ACME.ParametricNonLinEq{ACME.var"#108#122"{Matrix{Float64}, ACME.var"#106#120"{Vector{Float64}, ACME.CircuitNLFunc{Tuple{ACME.var"#72#73"{StaticArrays.SVector{2, Int64}, ACME.var"#23#24"{Float64, Int64}}}}}}, typeof(ACME.default_set_p), typeof(ACME.default_calc_Jp), Tuple{Vector{Float64}, Matrix{Float64}}}, initial_p::Vector{Float64}, initial_z::Vector{Float64})
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\solvers.jl:175
   [15] HomotopySolver{SimpleSolver}(nleq::ACME.ParametricNonLinEq{ACME.var"#108#122"{Matrix{Float64}, ACME.var"#106#120"{Vector{Float64}, ACME.CircuitNLFunc{Tuple{ACME.var"#72#73"{StaticArrays.SVector{2, Int64}, ACME.var"#23#24"{Float64, Int64}}}}}}, typeof(ACME.default_set_p), typeof(ACME.default_calc_Jp), Tuple{Vector{Float64}, Matrix{Float64}}}, initial_p::Vector{Float64}, initial_z::Vector{Float64})
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\solvers.jl:256
   [16] initial_solution(init_nl_eq_func::Function, q0::Vector{Float64}, nn::Int64)
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\ACME.jl:433
   [17] DiscreteModel(circ::Circuit, t::Int64, ::Type{HomotopySolver{CachingSolver{SimpleSolver}}}; decompose_nonlinearity::Bool)
      @ ACME C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\ACME.jl:198
   [18] DiscreteModel (repeats 2 times)
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\src\ACME.jl:150 [inlined]
   [19] macro expansion
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\test\runtests.jl:74 [inlined]
   [20] macro expansion
      @ C:\Users\sadguitarius\AppData\Local\Programs\Julia-1.7.2\share\julia\stdlib\v1.7\Test\src\Test.jl:1283 [inlined]
   [21] macro expansion
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\test\runtests.jl:62 [inlined]
   [22] macro expansion
      @ C:\Users\sadguitarius\AppData\Local\Programs\Julia-1.7.2\share\julia\stdlib\v1.7\Test\src\Test.jl:1283 [inlined]
   [23] top-level scope
      @ C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\test\runtests.jl:45
   [24] include(fname::String)
      @ Base.MainInclude .\client.jl:451
   [25] top-level scope
      @ none:6
   [26] eval
      @ .\boot.jl:373 [inlined]
   [27] exec_options(opts::Base.JLOptions)
      @ Base .\client.jl:268
   [28] _start()
      @ Base .\client.jl:495
Test Summary:       | Pass  Error  Total
simple circuits     |    2      1      3
  empty circuit     |    1             1
  only one resistor |    1             1
  resistor-diode    |           1      1
ERROR: LoadError: Some tests did not pass: 2 passed, 0 failed, 1 errored, 0 broken.
in expression starting at C:\Users\sadguitarius\.julia\packages\ACME\UHz8S\test\runtests.jl:44
ERROR: Package ACME errored during testing
```

martinholters, 2022-04-07:
Tests pass for me on Julia 1.7.2. As the error you have posted seems to originate from StaticArrays, maybe you need to update that? If you still have problems, could you share your Maniest.toml so that I can try to reproduce?

sadguitarius, 2022-04-07:
I think I figured it out. I had been trying to get these [ADAA examples](https://github.com/martinholters/ADAA_Examples.jl) working, and it looks like it was that project that pinned StaticArrays to a version that's incompatible with Julia 1.7.2 . I think I'd somehow downgraded StaticArrays in my global packages so that the ACME tests failed. Lots to learn. :-)

Just tried again with a clean package tree and the tests worked fine. I can reproduce the failed tests by running them in a project that includes the ADAA_Examples.jl package.

sadguitarius, 2022-04-09:
Closing, as the issue is not with ACME.jl. Sorry for the noise!

closed 2022-04-09 by sadguitarius

martinholters, 2022-04-28:
Thanks for pointing out the issue, though. I've bumped the version bounds in ADAA_Examples now (https://github.com/martinholters/ADAA_Examples.jl/commit/4de7c8a4be7f3ce5c662f1b40ea9314a727f3be5), so they should work out of the box with newer Julia, too.

sadguitarius, 2022-05-02:
Good to know, thank you!