HSU-ANT/ACME.jl #138: Define netnames based on external variables?

Arrondissement5etDemi, 2025-02-19:
Hi!

I would like to have netname for pins that is a function of external variables, such as 

```julia
for i = 1:10, 
    :(v_$i) = voltagesource(10), [+] == "vsrc_$i", [-] == gnd
end
```

Currently the code will show that i is undefined due to the QuoteNode nature of the pins.

Thanks a lot!!

martinholters, 2025-02-20:
This lacks a bit of context. Is that supposed to work in `@circuit` macro? That doesn't support `for` loops in the first place. And it's not meant for such advanced uses. You achieve what (i think) you want like so:
```
circ = Circuit()
for i = 1:10
    add!(circ, Symbol("v_$i"), voltagesource())
    connect!(circ, (Symbol("v_$i"), +), Symbol("vsrc_$i"))
    connect!(circ, (Symbol("v_$i"), -), Symbol("gnd"))
end
```
Also see the example at the very end of [the documentation section on circuit desciption](https://hsu-ant.github.io/ACME.jl/dev/ug/#Circuit-Description) and note that you can create `circ` with the `@circuit` macro to populate it with the basic components and then still add to it programmatically.

Does this answer your question/solve your problem, or did I isunderstand?

Arrondissement5etDemi, 2025-02-20:
Hi, yes I meant a bit of a pseudocode by writing a for loop. I see that it must be defined outside of the macro. I need the pins to have names that depend on function outputs, so I think your code here will work! Will try this out and let you know if it solves my problem. Thanks!!

Arrondissement5etDemi, 2025-02-24:
It works. Thank you!

closed 2025-03-12 by martinholters